101100/pca9685 #7: Version 2.2.0 appears broken

nkolban, 2016-06-21:
I installed 2.2.0 from npm and when I try and use it as a dependency in a project, I get the error:

```
Error: Cannot find module 'pca9685'
```
As I tried to diagnose, I find that the pca9685 module's package.json has `main` listed as `dist/index.js` but there is no such directory/file.

101100, 2016-06-22:
@nkolban Thanks for pointing that out!  Because of the change in how I was building, I needed to add a `.npmignore` file so `npm` wouldn't use `.gitignore`.  Can you give 2.2.2 a try to make sure it is working for you?  If so, I'll close the issue.

nkolban, 2016-06-22:
Awesome fast response.  Thank you 101100.  I'll test this evening after work and close the issue.

nkolban, 2016-06-23:
I downloaded 2.2.0 and all installed cleanly.  I haven't had a chance to put the functions through their paces but will assume all is good.  Many thanks my friend.

closed 2016-06-23 by nkolban