101100/pca9685 #8: Enhancement: Provide a speed control

nkolban, 2016-06-22:
Enhancement request:  In my project, I want the servos to move at less than the maximum possible speed.  For example, if today going from 0 to 180 degrees takes (and I'm making this up) 1.5 seconds, this would be a rate of 8.3msecs/degree.  I would like to be able to change that rate to other values ... for example if I made it 20msecs/degree then a 180 degree rotation would take 3.6 seconds.

Sadly, I haven't studied enough on the pca9685 or this module's internals to say if such a thing is easy, hard or even technically achievable.

101100, 2016-06-22:
The way servos work is that the signal indicates their destination, e.g. "turn to 180 degrees".  The speed is controlled by the servo itself.  To turn slower, you'd need to repeatedly tell it to turn part of the way.  Something like the example `servo.js` where you provide a range of values and use a timer to move to the next one slowly.  If you feel like implementing this in a general way (something with a `start` and `end` position, number of "steps" and total time to spread it over), I would accept a pull request.

nkolban, 2016-06-22:
Thank you 101100 ... I'll look at the example.  Unfortunately I don't have skills in TypeScript ... I'll look at the sample in detail and see if I can use that as the basis for a higher level consumer/wrapper of the module.

closed 2016-06-22 by nkolban