101100/pca9685 #14: Error initializing PCA9685

Wildvuur, 2017-04-30:
hi,
when i run the example servo.js i get this error:
```
  pca9685 Reseting PCA9685 +0ms
Error initializing PCA9685
```
when i copy paste the example under usage from the https://github.com/101100/pca9685 i get a positive output
```
Initialization done
Pulse range set.
Channel 6 is off.
```
i started looking in these examples because i get this Error initializing PCA9685 in node-red-contrib-pca9685 aswel

any ideas?

101100, 2017-05-02:
Can you please either paste in the code for both cases or add Gist links?  If one works and another doesn't it seems likely that one of the configuration flags is wrong.  That could be the port, device name, etc.

Another possible issue is permissions.  Are you running both examples as `root` or did you set up the device permissions for your user?

Do you have a servo attached or are you using it for LEDs?  Either way, try adjusting the example that works so that only the thing you need (pulse width or duty cycle) is set to the correct channel.

Wildvuur, 2017-05-02:
Jason,
it was late and i must have overlooked the address... sorry for this, i ran the servo.js again and it works like a charm now.
gues i will have to check with the person who wrote the node-red-contrib-pca9685 node,
i get the error i reported here, that how i started digging.
Sorry again for the trouble!

Matthieu

closed 2017-05-02 by Wildvuur