101100/pca9685 #15: LEDs flicker due to PCA9685 load balancing.

zjasper, 2019-06-02:
I'm getting a lot of flickering when multiple channels have the same off-step.

The PCA9685 uses a load balancing strategy when multiple channels have the same off step. The strategy involves turning all channels to of for a single cycle. [This library](https://github.com/NachtRaveVL/PCA9685-Arduino) can be modified to disable the load balancing by changing the PCA9685 object declaration from:

`PCA9685 pwmController;`
 to 
`PCA9685 pwmController(wire, PCA9685_PhaseBalancer_None);`

Disabling the load balancing does seem to clean up the flicker.

I'm rather new to this or I would've tried to figure out how to implement it myself, but is it possible to disable the load balancer in this library?

101100, 2019-06-02:
This library doesn't do load balancing, so what the other library is doing when `PCA9685_PhaseBalancer_None` is set is the same as the default of 0 for `onStep` that this library uses.  For example, here is the documentation for `setDutyCycle`:

https://github.com/101100/pca9685/blob/37f14d08502848fa916f984c19a87eb1c7ccaa28/src/pca9685.ts#L277-L290

If you are experiencing flickering, you might want to increase the frequency when you initialize the library from the default in the demos of 50 to something higher like 200.

101100, 2019-06-02:
If you'd like some help with your code, feel free to include it or a link to it here.

zjasper, 2019-07-12:
Sorry for the unacceptably long delay in this response.

I found the problem with my project, I was attempting to update the driver much faster than my rpi could process it. I believe my issues where related to the speed I was able to run instructions on the cpu.

I was using an original pi. First release.

I just bought a rpi 4 and it doesn't have the same issue. 

Thanks for your help and offer for more help!

101100, 2019-07-12:
Glad you solved your issue.

closed 2019-07-12 by 101100