101100/pca9685 #16: Don't find sample for continuous rotating servos

pedefe, 2020-05-30:
Hello,
I want to drive continuous rotating servo (for wheel) in "NodeJs" with "pca9685" on RaspBerry Pi. I succeed to drive "angle servo" with command pwm.setPulseLength( port, pulseLen, 0);
I see other commands like :
pwm.setDutyCycle( port, ratio, 0);
or
pwm.setPulseRange( port, min, max);
But I don't know how to drive the continuous rotating servo, 
- in one way and in the opposite way; 
- from null speed to max speed;
Can you give me sample ?
Best regards.

101100, 2020-05-31:
I've never used a continuous rotation servo before, but was able to find [this](https://www.robotc.net/wikiarchive/Tutorials/Arduino_Projects/Mobile_Robotics/BoeBot/Continuous_Rotation_Servo_Intro) reference that had this table:

| Input Pulse (ms) | Rotation Speed (%) | Direction of rotation |
|------------------|--------------------|-----------------------|
|  1.0             |                100 | Clockwise     |
|  1.1             |                 80 | Clockwise     |
|  1.2             |                 60 | Clockwise     |
|  1.3             |                 40 | Clockwise     |
|  1.4             |                 20 | Clockwise     |
|  1.5             |                  0 | N/A     |
|  1.6             |                 20 | Counter-Clockwise     |
|  1.7             |                 40 | Counter-Clockwise     |
|  1.8             |                 60 | Counter-Clockwise     |
|  1.9             |                 80 | Counter-Clockwise     |
|  2.0             |                100 | Counter-Clockwise      |

You should be able to use `setPulseLength` which takes microseconds (multiplying the values in the table above by 1000) for your continuous rotation servo.  If it was on channel 2, then `pwm.setPulseLength(2, 1000);` should go clockwise as full speed, `pwm.setPulseLength(2, 1500);` should stop and `pwm.setPulseLength(2, 2000);` should go counter-clockwise at full speed.  You should also be able to use values in-between for not full speed.

Hope this helps!

pedefe, 2020-06-04:
Hello,
Thank you very much, I will test it. 
But it seems to be specific for each product.
I have tried step by step, and the medium value for null speed is more around 1650.
The problem is to find the extreme values for max_clockWise and max_Counter-clockWise, because it is hard to see if the servo runs quicker or stay on same speed. Thank you.
Best regards.

101100, 2020-06-04:
@pedefe Thanks for letting me know about some of the details.  If you have a model number for your servo you might be able to look up the values somewhere.  Good luck!

101100, 2020-06-04:
You might want to try starting at a lower value and moving up until it comes on in reverse.  Then use that and the middle value to guess where the highest value would be.  E.g. if the first time it turns on is at 950 and your stopping point is 1650, the full speed reverse value might be 1650 + (1650-950).  Just an idea to try.

101100, 2021-02-12:
Closing this issue for now as there have been no replies.

closed 2021-02-12 by 101100