101100/pca9685 #21: About ESC

Canees, 2021-11-11:
Hello, can I use it to control the ESC?   , because they are PWM

101100, 2021-11-11:
I found that for my servos and speed controllers, they go from about 1300 to 1700 milliseconds, hence the reason the example has this:

https://github.com/101100/pca9685/blob/07dd57c72a376261d53c6ce4b2c74ae474a8db71/examples/servo.ts#L27-L29

I did start looking into a speed controller.  You need to start with the middle value of 1500 ms and hold it until the speed controller initializes (mine had two sequence of beeps, one when it turnd on and one when it was ready to go).  Then you would use the values less than 1500 to go in reverse (or brake) and the numbers greater than 1500 to go forward.  You'll have to experiment to find out what the upper range of speed is.

Canees, 2021-11-13:
> I found that for my servos and speed controllers, they go from about 1300 to 1700 milliseconds, hence the reason the example has this:
> 
> https://github.com/101100/pca9685/blob/07dd57c72a376261d53c6ce4b2c74ae474a8db71/examples/servo.ts#L27-L29
> 
> I did start looking into a speed controller for a car. You need to start with the middle value of 1500 ms and hold it until the speed controller initializes (mine had two sequence of beeps, one when it turned on and one when it was ready to go). Then you would use the values less than 1500 to go in reverse (or brake) and the numbers greater than 1500 to go forward. You'll have to experiment to find out what the upper range of speed is.
> 
> If it is for a drone or other one-way motor, you likely need to start at the low end and hold it for it to initialize. Think of the position the actual controller must start in when you turn on the device.

Thank you for your reply. I used the script given by the manufacturer for testing, and found that it can drive ESC. The range is within the range of 500~1500~2500 as you said, and the forward and backward steps are also calculated by adding and subtracting from 1500. Compared with using PWM of RSPI itself to drive the stroke, I will test your program later, and HOPE it can be used as well.

Canees, 2021-11-13:
> I found that for my servos and speed controllers, they go from about 1300 to 1700 milliseconds, hence the reason the example has this: [...]

Thank you for your reply. I used the script given by the manufacturer for testing, and found that it can drive ESC. The range is within the range of 500~1500~2500 as you said, and the forward and backward steps are also calculated by adding and subtracting from 1500. Compared with using PWM of RSPI itself to drive the stroke, I will test your program later, and HOPE it can be used as well.

Canees, 2021-11-13:
> I found that for my servos and speed controllers, they go from about 1300 to 1700 milliseconds, hence the reason the example has this: [...]

I changed the time to 20us, recursively it actually works。
ep:
/*
 * examples/servo.js
 * https://github.com/101100/pca9685
 *
 * Example to turn a servo motor in a loop.
 * Javascript version.
 *
 * Copyright (c) 2015-2016 Jason Heard
 * Licensed under the MIT license.
 */

"use strict";

var i2cBus = require("i2c-bus");

var Pca9685Driver = require("pca9685").Pca9685Driver;

// PCA9685 options
var options = {
    i2c: i2cBus.openSync(1),
    address: 0x40,
    frequency: 50,
    debug: true
};

// // pulse lengths in microseconds (theoretically, 1.5 ms
// // is the middle of a typical servo's range)
// var pulseLengths = [500, 1500, 2500];
var steeringChannel = 3;

// variables used in servoLoop
var pwm;
var nextPulse = 1500;
var step = 5
var timer;

// loop to cycle through pulse lengths
function servoLoop () {
    nextPulse += step
    if (nextPulse >= 2500) {
        step = -5
    }
    if (nextPulse <= 500) {
        step = 5
    }
    pwm.setPulseLength(steeringChannel, nextPulse);
    timer = setTimeout(servoLoop, 0.02);
}

// // set-up CTRL-C with graceful shutdown
process.on("SIGINT", function () {
    console.log("\nGracefully shutting down from SIGINT (Ctrl-C)");
    if (timer) {
        clearTimeout(timer);
        timer = null;
    }
    pwm.dispose();
});

// // initialize PCA9685 and start loop once initialized
pwm = new Pca9685Driver(options, function startLoop (err) {
    if (err) {
        console.error("Error initializing PCA9685");
        process.exit(-1);
    }

    console.log("Starting servo loop...");
    servoLoop();
});

101100, 2021-11-15:
Glad it is working.

closed 2021-11-30 by Canees